espressif/esp-idf-cxx #4: How to install

SinanAkkoyun, 2022-07-07:
Hi! Should one put the github repo in the idf_component.yml?
I can not find this awesome component in the component registry

Thank you for the great work

0xjakob, 2022-07-14:
@SinanAkkoyun We haven't "officially" release this yet. We plan to upload it into component registry once we have a state that is satisfying and probably not before IDF 5.0 is released. Furthermore, we just changed the default branch to `main`.

If you want to try it now, just clone it. Use the usual process for setting up the IDF environment (i.e. `. ./export.sh` in IDF repo), then you can go to the examples in or unit tests in this repository and try them right away. The component manager is set up for these to find the `esp-idf-cxx` locally in the parent folder. Please keep in mind that some structural parts may still change in the near future until we have a state to release.

0xjakob, 2022-07-27:
@SinanAkkoyun We added some information to the README. Please let us know if there's something else we can do within the scope of this issue. Otherwise we would like to close it in a few days.

Please feel free to continue discovering this repository and let us know if something doesn't work.

SinanAkkoyun, 2022-07-28:
> @SinanAkkoyun We added some information to the README. Please let us know if there's something else we can do within the scope of this issue. Otherwise we would like to close it in a few days.
> 
> Please feel free to continue discovering this repository and let us know if something doesn't work.

Yes of course! Thank you SO MUCH for the awesome cxx integration, developing on the ESP platform makes super fun that way.

If it's okay you can also delete this issue, for fewer issue count as it's not needed anymore :)

closed 2022-07-28 by SinanAkkoyun

0xjakob, 2022-07-29:
@SinanAkkoyun We're happy to hear that!