espressif/esp-idf-cxx #17: use fmt library for C++ logging (IDFGH-8779)

ljden, 2022-11-21:
### Is your feature request related to a problem?

More powerful formatting for log messages

### Describe the solution you'd like.

If using C++11/14/17: https://github.com/fmtlib/fmt
For C++20 it is part of the standard https://en.cppreference.com/w/cpp/utility/format

### Additional context.

This would only work for C++

ljden, 2022-11-22:
Might also be worth considering using spdlog for logging

SoucheSouche, 2022-11-23:
Hi @ljden,

You can already use the [fmt library](https://components.espressif.com/components/espressif/fmt) by including it into your project as a component using the [IDF component manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html).

closed 2022-11-23 by SoucheSouche

ljden, 2022-11-23:
@SoucheSouche this is not currently possible, and @igrr has provided a good example of what I'm after. Currently to use fmt with logging you need to construct a string, extract the c_str, and pass it to esp logger as `"%s"`. Having native support for fmt will allow for much more performant code without the need to construct strings or have a preallocated buffer for logging

ljden, 2023-01-03:
Just to add an extra use case for this, this would also mean string_views can be used in logging without needing to copy them. The current logging cannot do this as printf requires null terminated strings and string_views cannot guarantee they are null terminated

erhankur, 2023-01-06:
> Might also be worth considering using spdlog for logging

Just a basic example of `spdlog` on esp-idf.  https://github.com/erhankur/esp32-spdlog

ljden, 2023-01-08:
Oh thanks @erhankur! I wasn't able to get spdlog compiling - I must have been doing something wrong...

X-Ryl669, 2024-06-21:
> The current logging cannot do this as printf requires null terminated strings and string_views cannot guarantee they are null terminated

I'm using `string_view` just fine with printf like C function. Simply use `"%.*s"` for the syntax, and `str.length(), str.data()` in the argument list. I'm using it for string where I already know their length just to avoid a useless strlen computation in the runtime code.