espressif/esp-idf-cxx #19: Naming convention in GPIO

CyrilB1531, 2023-03-19:
All classes defined in gpio_cxx uses the following naming convention:
GPIO<Function>
but Output and OpenDrain use
GPIO_<Function>

As the project follows esp-idf coding style, these classes should be GPIOOutput and GPIOOpenDrain

0xjakob, 2023-03-23:
@CyrilB1531 We appreciate your sense for detail and noticing the diversion from the IDF style-guide. In this case, however, the diversion has been a conscious decision that we made to increase the readability of the name. `GPIO_Output` is more readable than `GPIOOutput` in our opinion.

Among the goals of the style-guide are readability and consistency. In rare cases like this, these two goals collide and a decision has to be made what is more important.

CyrilB1531, 2023-03-23:
Ok,
i will close the issue then

closed 2023-03-23 by CyrilB1531

0xjakob, 2023-03-27:
@CyrilB1531 We made this explicit in the [IDF code style guide](https://github.com/espressif/esp-idf/blob/master/docs/en/contribute/style-guide.rst). The change should be visible when the IDF master branch is updated the next time.