espressif/esp-idf-cxx #29: 9esp_timer_is_active ?

diplfranzhoepfinger, 2024-04-07:
Hi, 

in the Timer Wrapper, 

we cannot ask if a Timer is active ? 

[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#_CPPv419esp_timer_is_active18esp_timer_handle_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#_CPPv419esp_timer_is_active18esp_timer_handle_t)

can this be added somehow ?

0xjakob, 2024-04-12:
@diplfranzhoepfinger Yes, this can be added. We will add something like the code in the following patch: 
[0001-WIP-feat-add-is_active.zip](https://github.com/espressif/esp-idf-cxx/files/14961492/0001-WIP-feat-add-is_active.zip)

diplfranzhoepfinger, 2024-11-06:
cool.

diplfranzhoepfinger, 2026-09-22:
Opened a PR for this: espressif/esp-idf-cxx#46

@0xjakob had already put together a WIP patch adding `is_active()` back in April 2024 — I picked it up and applied it, tagging you here in case you'd like to weigh in.